SimonRafferty/Webasto-Heater---Replacement-Controller #3: Voltage regulator type

prensel, 2022-02-06:
What brand/type of 5V voltage regulator is used ? 
I have a hard time finding one at RS with the same dimensions..

SimonRafferty, 2022-02-06:
It's actually just one from eBay - there are hundreds of little Chinese switching supplies.  
Search for 5V Buck Converter.  For example: https://www.ebay.co.uk/itm/273157036523 
This is probably the one I've used as I've got loads of them.  They're tiny and you can select the voltage with a solder bridge between two pads.

prensel, 2022-02-06:
Ok thanks, unfortunally the one you refer to has a different layout of the pins. 
 I was hoping I could use a drop-in item

SimonRafferty, 2022-02-06:
The holes in the board were for another type which turned out to be useless.  The one I'm using just has wire links between the two.  Not as tidy, but thats less important than it working!

closed 2022-07-02 by SimonRafferty