SimonRafferty/Webasto-Heater---Replacement-Controller #4: Altitude adjustment?

Ramdough, 2022-07-02:
Any thoughts on having altitude adjustments to your fuel ratios?   

Maybe add a pressure sensor?

SimonRafferty, 2022-07-02:
It's definitely an option!
Have you found that the original Webasto controller struggles at high altitude?

I wonder if, when the air is thinner, there is less resistance to the cumbustion fan spinning (as it's power controlled, rather than rpm), allowing it to spin faster and allow it to push more volume of air through, effectively self regulating the oxygen passing through the combustion chamber?  That's unlike an engine where the volume of air raken in per cycle is fixed by the displacement.

I'd be interested to know how they cope with differing altitude?

Ramdough, 2022-07-06:
[image]

Ramdough, 2022-07-06:
[High Altitude Compensator Manual 10-2011.pdf](https://github.com/SimonRafferty/Webasto-Heater---Replacement-Controller/files/9056373/High.Altitude.Compensator.Manual.10-2011.pdf)

lbartik, 2022-07-15:
> It's definitely an option! Have you found that the original Webasto controller struggles at high altitude?
> 
> I wonder if, when the air is thinner, there is less resistance to the cumbustion fan spinning (as it's power controlled, rather than rpm), allowing it to spin faster and allow it to push more volume of air through, effectively self regulating the oxygen passing through the combustion chamber? That's unlike an engine where the volume of air taken in per cycle is fixed by the displacement.
> 
> I'd be interested to know how they cope with differing altitude?

Very cool project.  

Wouldn't that be convenient if there was some "natural compensation" of a power-governed fan as elevation increases (density decreases).  Fan RPM does increase, thus delivering a slightly greater volume of lower density air at fixed power.  However, the relationship between mass flow rate and density is linear-proportional, as is the relationship between rpm and volumetric flow, whereas the relationship between fan power and rpm is cubic.  The net result is the mass flow ratio of a fixed power fan at two different air densities is equal to the density ratio raised to the 2/3 power.     m-dot2/m-dot1 = (rho2/rho1)^2/3.   For example, altitude ~10000ft, density ratio is 0.70 of sea-level.  Mass flow of fan is 0.7^2/3 = 79% of sea-level mass flow rate.   I think the real fan and brushed DC motor probably suffer further inefficiency at higher RPM and static pressure as you move around their combined efficiency curves.

I looked at Eberspaecher's high-altitude scheme for the newer S3 B5S hydronic heater with CANbus and found it is simply an on/off setting (on the Easystart PRO LCD screen) which must limit the maximum fuel injection rate, raising the maximum safe operating altitude ceiling some discrete amount.  A baro sensor accessory was available for older models but they must have found the on/off lean-burn switch to suffice or just haven't got around to updating that box to be compatible with these newer CANbus controllers.

In HVAC terms, you are playing with the boiler's turndown ratio, the ratio of the boiler's minimum capacity to maximum capacity.  Ramjob makes a good point, there is such a thing as modulating too low.  I don't pretend to know all the risks but combustion chamber sooting seems to be the primary one.  Though it sounds backwards, this must be a greater risk with gasoline boilers.  The turndown ratio for diesel ESPAR models is about 26% where the gasoline turndown ratio is 36% (per spec sheets).  Perhaps another risk of modulating too low is condensing flue gas in the combustion chamber.

I have about 5 gallons of coolant for thermal capacitance in my van.  It short-cycles when the outdoor temperature is warmer and the heat load is low.  I liked the notion of the boiler running more frequently at full output to de-carbonize, but I haven't considered the energy expended lighting it back up every cycle.  It does eat some battery.  Thanks for sharing.  If you have documented the physical installation anywhere on the net, I'd like to see it too!

SimonRafferty, 2022-07-16:
> If you have documented the physical installation anywhere on the net, I'd like to see it too!

I've written up most of the van build here: https://forums.lr4x4.com/topic/109170-lockdown-camper-van-build-not-even-slightly-lr-but-potentially-useful-bits/#comments

There's only a little bit of detail on the heater - more text than pictures though.

closed 2022-12-22 by SimonRafferty