SimonRafferty/Webasto-Heater---Replacement-Controller #5: flame control

softwarecrash, 2022-11-19:
hey, interessting project,
but one question, why you not use the glow plug as flame control like the orginal?

SimonRafferty, 2022-11-19:
Originally, because measuring the difference between 0.3 & 0.6 Ohm is relatively hard to do simply.  

I tried just using a shunt resistor & measuring the voltage across the shunt with an ADC - but the signal is right down in the noise, even with an Op-Amp & it was unreliable.

I have a version in the works which uses a hall effect current sensor, then pulses the power to the Glow plug for 10mS to measure the current flow, before it has a chance to heat up too much.  
I think the OE board uses a current mirror to measure in the temperature in the same way.

This has stalled at the moment because of the apparent chip shortage.  When I came to make the board - they were not available until next year.  I could re-design the board to take a sensor that is available - but I wasn't in a hurry so figured I'd wait.

Watch this space - as they say!

softwarecrash, 2022-11-19:
i hope you can get the glow plug up to work as termistor, so a new pcb design that can swapped with the oem board without extra sensors sound good.
so in pcb and hardware design i cant help, but when you have done that i will look that i can help you with the code. you use a esp32? so i i can try to make a webinetrface and mqtt connection (have heard from camper vans that have running a smarthome system onboard) so the heater let it integrate.
im self have the heater as staionary garden house heating solution, but actual the pump get stuck when it was could

SimonRafferty, 2022-11-19:
I would guess, looking at the photo, they are using the 1R0 resistor (Left middle) as a shunt in series with the glow plug & measuring the voltage across it, through an Op-Amp.  Their design is a lot more 'polished' than mine - so maybe they have a lower noise floor & can measure the signal reliably.

I was using an ESP32 but changed to an M0 because a feature of the ESP is the WiFi interferes with the ADC readings (one of the two ADC converters mainly).  I did build a reasonable Android App to control it.

I used the M0 because it was what I had laying around and it plugs into the same socket as the ESP - so I could make it work without changing the PCB.  I should really go back to the ESP & use the other ADC channel - then re-establish the App & a web interface.  A job for another day though!

Thanks for the offer - I imagine you could make a way better job of a web interface than me!  My app looked rather 'Agricultural' ;-).

softwarecrash, 2022-11-20:
a little thing i have noticed in your fan.ino
in the comments your write the fan runs at 10V 
i think thats not so correct, the fan was powered by a high frequenzy pwm, and runn at heater full load at 80-93% deepend on the co2 calibration, by hex80 calibration the fan will run at 83% so that is like 10V 
so when the heater goes to cooldown and blow out the flame the fan goes to 100% duty cycle
so the fan speed corespondign with the pump frequency, and co2 calibration

SimonRafferty, 2022-11-20:
You're right!
On the original processor, the analogWrite was 10 bit - so the comment '// Assuming a 13V Input, 769 gives 10V' was about right.
On the current version, it's 8 bit so the mapping: "mapf(fan_speed, 0, 100, 0, 239)" actually gives about 12V rather than 10V.

It was clearly an error.  Then I calibrated the fuel & fan based on the error - so the fueling is about right.

When I have time, I'll go through the code & re-calibrate the fan.  I think the highest voltage tha fan will see is about 10.8V - but it really ought to be 10V.

Thanks for pointing out the error!

softwarecrash, 2022-11-20:
fu*** cold outside, i have quick check it, in the working phase the fan will run around 10V, but in the blowout phase it will get fullspeed at 12V. 
its simply a (brushed?) dc motor, i think it cant damage at 12V.
so when you have a cooldown/blowout phase implemented you can run the fan at full power for faster cooldown or blof off the flame.

SimonRafferty, 2022-11-20:
It is just a DC PM Motor.  Running at higher voltages for short durations is not a problem.

softwarecrash, 2022-11-20:
jep, the fan motor will permanently run at 11-14volts without any damage.

so the idea will not go out of my head to watch the flame like the oem, so sorry im not a electrical pro, so i think with a cricuit like this (https://www.electronics-lab.com/project/arduino-based-milliohm-meter-lcd-display/) that will switch on measuring when the glow plug switched off is a base? or a simple way a lm317 to get variable voltage from a resistor for measuring with ADC

mlynekg, 2022-12-04:
Hello
Congratulations on a great idea.
Wouldn't it be possible to add support via a web browser to the project like Ray Jones did http://www.mrjones.id.au/afterburner/

Regards

SimonRafferty, 2022-12-04:
@mlynekg - The afterburner is really neat!

The first version of this had a web based control interface.  I'd originally used an ESP32 - which made it fairly easy to implement.
Unfortunately, I had to move to an M0 for other reasons & it has no WiFi or Bluetooth.

It's still on my ToDo list - but there are more important things to address first.

I currently use an eWeLink, 4 relay board https://www.ebay.co.uk/itm/354419373377  to control mine.  One relay for heater on/off, one for the underfloor heating pump & one for the convection blower.  It works pretty well.

I would, one day, like to make another version with a web app to control it, along with diagnostics & over the air updates.  Watch this space - as they say!

SimonRafferty, 2022-12-20:
The ACS711 Current sensors turned up, so I built up a board.  This is the results from the first test run.  It switches on the glow plug for 10s then starts measuring:
Batt V=12.65| Glow A=20.34| Glow R=0.62| Glow T=112.44
Batt V=12.87| Glow A=21.22| Glow R=0.61| Glow T=104.79
Batt V=12.87| Glow A=20.92| Glow R=0.61| Glow T=108.90
Batt V=12.88| Glow A=21.70| Glow R=0.59| Glow T=98.06
Batt V=12.89| Glow A=22.56| Glow R=0.57| Glow T=86.93
Batt V=12.90| Glow A=21.36| Glow R=0.60| Glow T=103.34
Batt V=12.89| Glow A=22.71| Glow R=0.57| Glow T=85.06
Batt V=12.91| Glow A=23.73| Glow R=0.54| Glow T=72.88
Batt V=12.91| Glow A=23.54| Glow R=0.55| Glow T=75.17
Batt V=12.90| Glow A=23.14| Glow R=0.56| Glow T=79.77
Batt V=12.91| Glow A=23.05| Glow R=0.56| Glow T=81.06
Batt V=12.91| Glow A=25.22| Glow R=0.51| Glow T=56.61
Batt V=12.92| Glow A=25.81| Glow R=0.50| Glow T=50.95
Batt V=12.91| Glow A=22.71| Glow R=0.57| Glow T=85.51
Batt V=12.92| Glow A=26.10| Glow R=0.49| Glow T=48.11
Batt V=12.91| Glow A=25.73| Glow R=0.50| Glow T=51.60
Batt V=12.92| Glow A=24.32| Glow R=0.53| Glow T=66.49
Batt V=12.92| Glow A=23.90| Glow R=0.54| Glow T=71.16
Batt V=12.93| Glow A=26.27| Glow R=0.49| Glow T=46.78
Batt V=12.96| Glow A=23.85| Glow R=0.54| Glow T=72.65
Batt V=12.91| Glow A=26.71| Glow R=0.48| Glow T=42.30
Batt V=12.94| Glow A=25.27| Glow R=0.51| Glow T=56.82
Batt V=12.93| Glow A=26.15| Glow R=0.49| Glow T=47.87
Batt V=12.93| Glow A=26.22| Glow R=0.49| Glow T=47.17
Batt V=12.92| Glow A=25.73| Glow R=0.50| Glow T=51.83
Batt V=12.94| Glow A=24.10| Glow R=0.54| Glow T=69.36
Batt V=12.92| Glow A=27.15| Glow R=0.48| Glow T=38.56
Batt V=12.94| Glow A=28.03| Glow R=0.46| Glow T=31.21

The absolute (calculated) temperature probably isn't accurate - but it was interesting to try.  At 20C, the glow plug measured 0.45 Ohms (on a multimeter) so the readings are not far off.

Hopefully this week, I'll be able to test it with the heater running properly.  The above was a 'dry' test with only the glow plug connected.  Looks promising though!

softwarecrash, 2022-12-20:
> The ACS711 Current sensors turned up, so I built up a board. This is the results from the first test run. It switches on the glow plug for 10s then starts measuring: Batt V=12.65| Glow A=20.34| Glow [...]

looks great, that a big step to build a open replacement

SimonRafferty, 2022-12-21:
It's ALIVE!
After a bit of tweaking, the Flame Sensor / Exhaust temperature works well.  I discovered the Current sensor needed a bit of thermal compensation.  It's well within it's operating temperature range - but the read current reduces with temperature - enough to cause spurious readings.

I built a simple web server delivering the data to a page.  Unfortunately, I found it too slow.  When a client connects & requests a page, it takes a second or two - enough to miss several pulses of fuel.
In the short term, I've implemented Blynk which only takes 100mS or so & doesn't cause a problem.
I'm going to try making the fuel delivery controlled by an Interrupt Timer to see if that helps.

I've added a branch for the code.

SimonRafferty, 2022-12-22:
I've added a new version here https://github.com/SimonRafferty/Webasto-Heater---Replacement-Controller/blob/Webastardo-V3/README.md
Which includes compile time options for different versions of the PCB & whether you want to use WiFi / Blynk

With this version, you can swap your existing Feather M0 for a WiFi version - then use Blynk or a WiFi interface, even on a V1 or V2 PCB.

softwarecrash, 2022-12-22:
> It's ALIVE!
> After a bit of tweaking, the Flame Sensor / Exhaust temperature works well.  I discovered the Current sensor needed a bit of thermal compensation.  It's well within it's operating [...]

The web Server blocking the working system of Pump etc? That realy Bad.
Let me Check later, so i must build a Code Version that i can Test without Hardware.

SimonRafferty, 2022-12-22:
If you want to give it a go with test hardware, ping me an email & I'll post you a V3.0 PCB?

softwarecrash, 2022-12-22:
sadly i cant build actual a hardware test setup, i must order some pices for a second heater, and electronic components.
my mail is softwarecrash )ath( hotmail dot com btw.. dit you know that the oem board checking the compustion fan for blocking or wire break? its simple power up the fan, cut off the power and measure the voltage generated by the motor, and power on again, is this easily to realize with a voltage divider?

let me check start next year that i can work on it without hardware first.
a short look at your wifi section, you use a verry basic wifi webserver routine with delay and awaiting answer and end, thats blocking all. so i hope i can put in the incredible fast asyncwebserver librays wich brings a wifi manager and other nice things like websocket.

so i prefer to use a esp82xx board for use, or better get the big brother esp32? later usable for bluetooth (never used but sound funny)

SimonRafferty, 2022-12-22:
I actually started the project using an ESP32 but had too many problems with the ADC conversion being interfered with by WiFi - even when the WiFi was not active or enabled!
The M0, even with WiFi, on the other hand just works!

Using something like a ProMicro with an ESP32 as a co-processor isn't a bad idea though.  I'll have a think on that option & see if I feel inspired to have a go!

softwarecrash, 2022-12-22:
> I actually started the project using an ESP32 but had too many problems with the ADC conversion being interfered with by WiFi - even when the WiFi was not active or enabled! The M0, even with WiFi, [...]

yes, i think its the securest way, so have often struggeling with the esp wifi and timing sensitive or dangerus devices. so when the esp goes wild, it dint burn the house :)

prensel, 2022-12-23:
I'm following this because I have built the hardware but lacking the exhaust temp sensor. Considering a current sensor has come to mind but then I have to change my current setup. Was rather looking into using the already present water temp sensor as indication for flame control. I assume (havent measured) the water temp would soon drop when there is no flame and the pump is running. So say at a deltaT of 5 degr I would stop pumping fuel and restart the unit.

SimonRafferty, 2022-12-23:
@prensel It would, depending on how you're using the heater.  In my case, I'm heating a volume of water which is continually circulated - so if the flame extinguishes, the water temperature sensor only cools at the rate the larger volume of water cools - which takes hours.
If you were using it to heat water on demand, continually feeding in cold water - then it could work.

Jonburford1, 2022-12-31:
Morning Simon. Really excited to hear about your developments, have you made the v3 board available yet? Wouldn’t mind getting going on one. Cheers!

SimonRafferty, 2022-12-31:
https://oshwlab.com/simonrafferty/webastardo-3-0

I've published this - and it works OK.  I am thinking about revising the whole board design though.  I designed the board based on the parts I had laying around, rather than ones which you can buy from JCLPCB as a made up board (which most people seem to want).

[image]
This is a trace of the temperature read through the glow plug.  The 'tepmerature' shown isn't accurate - but does show enough swing between flame & no flame to be useful / reliable.

I've been running this as a heater for my office (connected to a domestic radiator) for a couple of weeks and it's been pretty good!

Jonburford1, 2023-01-03:
Thanks Simon,
Good progress, I’ll hold fire for now then and see what happens. Let me know if there’s any testing you’d like me to help with.
Cheers.

Ramdough, 2023-01-03:
Simon,Do you think we could get a simple altimeter added on the next revision?   The circuit would be simple.   Next time you turn a board, I would be interested in pitching in on a few.   ThanksD

https://oshwlab.com/simonrafferty/webastardo-3-0

softwarecrash, 2023-01-03:
simon and me have a idea, so simon will think about it to split the prozessors, in this case i think its a better way to ad some altimeter and other things to the co prozessor with wifi that communicate with the main cpu

SimonRafferty, 2023-01-03:
I was actually thinking about the barometer.  Most barometer modules can measure temperature too.  I wonder if there's a module with sufficient pressure & temperature range to use if for both altitude and measuring the temperature of the body of the heater?
All those I've looked at so far top-out at 85C.  I would rather the measurement range extend at least to 100C.
Any ideas?

SimonRafferty, 2023-01-29:
I've uploaded a couple of small fixes to the version with the flame sensor - just in case you're thinking "#@:&% thing doesn't work!" ;-)

It was causing unreliable starts at some ambent temperatures.  However, that asside, this version seems to work pretty well.

zorrobyte, 2023-10-25:
@SimonRafferty Do you have details on how you implemented the ACS712 into your PCB design? I can't find the SMD on the board anywhere. Are you using a breakout ACS712 board?

SimonRafferty, 2023-10-25:
It's implemented on the V3.0 Board.  That's the only difference between 2.1 & 3.

However, you could easily implement it off the main board by interrupting the power feed to the glow plug & putting the ACS712 in series with the glow plug.  There are plenty of easy modules on eBay that will be easier to solder than an SMD device.

SimonRafferty, 2023-12-07:
V3.0 has been running nicely, heating my workshop, for almost a year now.  I've recently introduced a version controlled by Home Assistant https://github.com/SimonRafferty/Webasto-Heater---Replacement-Controller/tree/Webastardo-V3-WiFi-Home-Assistant

There have been a lot of requests for different features here, by email, Facebook etc.  I decided to design a 'universal' version which is built on a rectangular board (so it has to be in a separate box), using XT30 & XT60 connectors for power and screw terminals for everything else.  The board is mostly SMD, so you can get JCLPCB to assemble the majority of it.

It has 6 MOSFET outputs, all with the same current rating, so any can be used for the glow plug.  They all have current measurement capability - which can be used for fault detection as well as measuring the flame sensor.  I've also moved away from Thermistors and changed to K-Type Thermocouples which are accurate from -100 to +1000C and thus can be used for water / air / exhaust temperature.

To accomodate all the other requests, such as displays, LoRa remote control, Altimeters, GPIO etc, I've broken out one SPI Port and I2C so you can easily add peripherals.  It also uses an ESP32 and runs a web-server for control.  At the moment, the web server allows all the settings (fuel mixture for example) to be adjusted live, which will make tuning much easier.

The boards have not arrived yet - but once they have and I've tested them properly, I'll start a new branch.  This will get a significant re-write of the code.  Hopefully this will make it easier to understand!

softwarecrash, 2023-12-07:
> V3.0 has been running nicely, heating my workshop, for almost a year now. I've recently introduced a version controlled by Home Assistant [...]

wow... i dont have time to do some thing , so other projects needs my time, but what you make sounds verry good, looking forward to order a pcb to test it :)
thank your for your great work

Identitry, 2025-01-03:
> V3.0 has been running nicely, heating my workshop, for almost a year now. I've recently introduced a version controlled by Home Assistant [...]

Hi @SimonRafferty!
How is your new board processing?
I'm planning to build a campervan and want to include underfloor heating however I do need a controller like yours and I would love the latest version with ESP32, SenseMOS (?) and thermocouples.

Please let me know if you need any help to continue the project!?
If you would have anything to share like schematics and maybe some code I would love to take over where you left off?

SimonRafferty, 2025-01-05:
I've been too busy at work for most of the past year to move forward with this.  However, while I was still working on it, I came to the conclusion that Thermocouples were a problem.
Low cost thermocouples from China tend not to have isolated cases - there is leakage current from the thermocouple wires to the casing.  In some cases, the casing is connected to one of the thermocouple wires.
This leads to problems.  The first thermocouple interface chip I used, registered this as an error.  The next version didn't throw an error but the continuity led to high noise levels.  The motors used in Chinese heaters tend to generate a lot of noise too which compounds the issue.
I bought high quality (expensive) thermocouples from RS which worked perfectly.  Unfortunately, at £60 each x 3 it makes the controller unrealistically expensive.

I added a display to V4 just to tell you the IP address for the web interface (when it's connecting to a router with DHCP).

I had been running my V3, using the glow plug to detect exhaust temperature.  It works OK for a genuine webasto heater - but the glow plugs in Chinese heaters don't show a big enough change in resistance with temperature to read accurately.  I've eventually given up on that idea too!  The V3 which runs my office heating now uses a PTC100 thermistor and it's a whole lot more reliable.

I do plan to pick this up again when time permits.  The next version (V5) will probably be based on a TTGO T-Display ESP32, with a web interface and the display showing the IP Address and running metrics.  V5 will likely use PTC100 thermistors, more surface mount components - and a load of other changes based on what I've learned so far.  Unfortunately V4 has reached the end of the road and is not reliable enough to release.

closed 2025-01-05 by SimonRafferty

Identitry, 2025-01-05:
> I've been too busy at work for most of the past year to move forward with this. However, while I was still working on it, I came to the conclusion that Thermocouples were a problem. Low cost [...]

Guess I've forgot to thank you for the great job you've made so far @SimonRafferty!!!
Sad that V4 didn't turn out to work well but I'm glad you're sharing your experience cause I'm currently working on my own version (with a lot of help from your code, yes if released you'll get a lot of credit!!!). 

I'll be doing my design on the ESP32 C6 for being able to add Zigbee and maybe Matter in the future for control, the latest Matter specification v1.4 included a couple of new clusters that could be interesting especially the "Boiler". I guess it'll work on most ESP32's.

Currently I'm building a library for the logic handling of the heater, this is basically going to be a State Machine and the library will be isolated from the hardware with a hardware abstraction layer making the code "a bit" easier to understand and manage even though it requires some "abstract" stuff! :-)

Currently working on the State Machine because I couldn't find any one good enough but ChatGPT is your friend in this matter.

When and... if, I come far enough to publish it to a private repo I'll invite you for a review, feel free to join!

SimonRafferty, 2025-01-05:
I really ought to publish the V4 code - but have resisted as the V4 hardware isn't good.  Although I've not implemented the FSM as a library (great idea by the way), I have re-written it a fair bit.
The most important change is to use RTOS to split the running of the FSM from updating the website and any other less critical tasks.  I recommend doing something similar as I could not make them play nicely with a single thread.

Identitry, 2025-01-05:
> I really ought to publish the V4 code - but have resisted as the V4 hardware isn't good. Although I've not implemented the FSM as a library (great idea by the way), I have re-written it a fair bit. [...]

That is my idea too, with lower priority for the nitty-gritty task(s)!
Will require some thread-safe handling but definitely worth it.

It would be nice if you could share V4 anyway, maybe it could help me on the way...

SimonRafferty, 2025-01-05:
I've now added a branch for V4.  The original FSM now lives in Heater.ino

Identitry, 2025-01-05:
> I've now added a branch for V4. The original FSM now lives in Heater.ino

Thanks a lot! The refactoring of control.ino and heater.ino, it will definitely help me...

My FSM is starting to look pretty good, some features:
*Context object with all variables, this object is also thought of being the object transferred over MQTT, etc. (public members).
*Four transition types, Manual, Timed, Event-based, and Automatic.
*Transition Guards, stops transition to state if guard returns false, automatic transitions makes transition when guard returns true. 
*Global state, all updates will pass this one before running the state code, a good place to sample temp sensors and update context object.
*Fluent API.

SimonRafferty, 2025-01-05:
I like the idea of a general purpose FSM library.  Useful for a lot more beyond this project.  

Passing objects over MQTT sounds like it has potential for brain-melting problems down the line ;-)

Identitry, 2025-01-05:
I see you still use ACS711/712 with your MOSFETs, have you seen the cool PROFETs with built in current (and voltage) sensing and advanced protection?
https://www.infineon.com/cms/en/product/power/smart-power-switches/high-side-switches/profet-plus-2-12v-automotive-smart-high-side-switch/

I'm thinking about using this one, not cheap (4-5$) but pretty smart and available with JLCPCB, it even says glowplug as one of its main use cases:
https://www.infineon.com/dgdl/Infineon-BTS70012-1ESP-DataSheet-v01_20-EN.pdf?fileId=5546d462758f5bd1017598473e2a39a7

SimonRafferty, 2025-01-05:
They look good.  However, one of the reasons I've stuck with discrete, through-hole MOSFETs is they're easy to replace when something goes wrong!