SimonRafferty/Webasto-Heater---Replacement-Controller #8: Working on a MicroPython fork of sorts, kind of working!

zorrobyte, 2023-10-26:
Excited, just got it semi-working in simulation:
https://wokwi.com/projects/379601065746814977

Repo:
https://github.com/zorrobyte/esp32-universal-diesel-heater-controller/tree/dev

I plan on basing this on the Webastardo hardware and possibly forking your PCB at some point (will be OSHW/OSS as all of this I'm doing). Lots of work need done for the startup/shutdown procedures and checking the main control loop. Also need to add a watchdog and safety checks to cut fuel if it crashes, etc.

Excited to collaborate in proxy. I'm thinking a LoRaWAN chip or something could be added to work with a standalone LCD Display for a control panel and temp probe. Lilygo epaper comes to mind for a low power solution that is battery powered.

P.S. know of any good diesel heater hackerspaces online? Discords, forums? Gonna ship my stuff to David to check out once I have something tangible.

SimonRafferty, 2023-10-26:
I don't think there any 'hackspaces' just a few individuals.   Its not a
bad idea though!

Im not keen on discord, but maybe redit or slack?  Maybe ask David too.

I think there's a lot of interest in hacking heaters, including the Chinese
ones.  The Afterburner code is OS - maybe he would like to join a group too.

There's also a guy, Roy Murkin. We collaborated to hack Eberspacher air
heaters control interface a few years ago.  That's what led me to build my
own controller.

I'm not a Python fan (grown up with strong type casting & it's difficult to
give up 🙂) However,  lots of people love it, so it would be a great
project fork.

It would make it a lot easier to develop a good web UI.

I do like LoRa (I'm a Radio Ham too).  LoRaWAN is good & I have a gateway
running - but still the coverage is poor.  Might be worth looking at SigFox
which is a commercial LoRaWAN network which has amazing coverage.  I use it
to track my Bike.

On the other hand WiFi probably makes more sense to more people.  Ive been
working on integrating with home assistant using ESPHome which gives the
same kind of remote access & monitoring but leaves the UI up to the end
user.

Are you in the UK?

Regards,

Si

b0ch3nski, 2023-10-27:
Hi folks,

I've put a watch on this repo and follow the Webastardo project for some time - my plan was to use RP2040 and TinyGo (oh hai, strong type casting) and hopefully one day I will have a time to make it happen :smile:

If you came up with a Discord/Slack/whatever for a Diesel heater hacking, please invite me too!

Greetings from Poland,
Piotr.

zorrobyte, 2023-10-27:
Simon,

David got ahold of me and invited me to his group for ECU hacking:
https://www.facebook.com/groups/1611135115986300

Ray kicked ECU convo out of the Afterburner FB group, which makes sense, custom ECU doesn't work with Afterburner and he's busy supporting his customers over there.

Roy Murkin, I'll do some googling!

Totally get the type casing argument, complied languages can also be faster out of the box!

ESPhome is great, use it with afterburner and other ESP32 projects. I plan on supporting MQTT itself to keep it simple for now. May look into a MicroPython HA library if one exists to make it easy.

I'm in the USA, hello! 👋 

Greetings, Pitor!

Feel free to join the FB group above!

TinyGo sounds fun and reliable! Be sure to put some software defined resistors/pots on your board as heater thermistors may differ. I get the time limitations for sure.

I plan on a serial console diag/config for now to keep it simple, eventually can come up with a webpage, other method. Always a good idea to keep it simple!

Pie in the sky (unlimited time and budget) standalone LCD that works wirelessly to the ECU, but one can dream. Need to get basics working first!