SimonRafferty/Webasto-Heater---Replacement-Controller #10: Petrol Thermo Top C possible?

menotuu, 2023-12-06:
Can this board work with a petrol based Thermo Top C too possibly?

SimonRafferty, 2023-12-07:
Short answer - Yes!
Longer answer - the air-fuel mixture will need to be edited.  So it will require a bit of experimentation.

The way I originally found the air-fuel mixture for the diesel version was to measure the fan speed (using an optical rpm meter with a painted white mark on the fan) and by timing the fuel-pump pulses using the stock controller.

For the startup routine, I copied this exactly.  Once the heater was running, I just needed the ratio of fan rpm to fuel pulses per minute.  So long as the ratio stays roughly the same, it will stay alight!

Then I used a cheapo Carbon Monoxide detector to tune the mixtures for the minimum CO.

menotuu, 2023-12-09:
I don't think that I need the wifi functions.
this should be enough or? https://www.dfrobot.com/product-2095.html

can you help me later after soldering the pcb to bring it to live with petrol?

the thing is, the stock controller is notable to run, because its VW canbus controlled.

menotuu, 2023-12-12:
and 3. are the pcb text prints reversed for whater therm and exhaust therm?
because the code says
int water_temp_pin = A1;        // exhaust temp? pcb text saysgoes to pin 3 
int exhaust_temp_pin = A2;      // water temp? pcb text says goes to pin 5

4. whats does Analog IP 03 mean? is this lambda_pin for future?

sorry I only want to understand it all correctly before burning hardware.

SimonRafferty, 2023-12-14:
On the board, there is a 6 pin connector - the same as the original Webasto board.

The Pin numbers refer to that, not the pins on the processor.

So, pin 3 on the connector goes to A1 on the microprocessor

closed 2023-12-14 by SimonRafferty

menotuu, 2023-12-14:
misunderstood.
Pin 3 goes to A1 via R11 4k7 divider for exhaust temp. right
And pin 5 goes to the A2 via R12 100k divider and is for exhaust pin too. 
I know, either this or this. because pcb text says use r11 or r12 divider.

But in the code A1 is for water temp sensor, and the text on the pcb on left side claims that too.