SimonRafferty/Webasto-Heater---Replacement-Controller #11: AC711 vs AC712 voltage

menotuu, 2024-01-08:
The AC711 has a Supply Voltage of 3min 3.3typ 5.5max V.
The AC712 has a Supply Voltage of 4.5min 5typ 5.5max V.

All of the IO pins of Adafruit Feather M0 Basic run at 3.3V.

on pcb there is a Ac712 printed but connected with 3.3V. how can this work?
in the sketch there is AC711 written.

I ask, because on my setup I never can bring up to glow the glow plug with AC712 supplied with 3.3V, and maybe its the fault on my setup.
but if I cut the 3.3v supply and supply the AC712  with the 5v from converter, the output would be too high for the IO pins of the Adafruit Feather M0 Basic??

Have you installed a AC711 or AC712?

SimonRafferty, 2024-01-08:
> ID: <SimonRafferty/Webasto-Heater---Replacement-Controller/issues/11@
> github.com>

I used an AC712.  I originally intended to use a 711, but they were not
available at the time due to Covid.

Despite the rating, I found it worked fine, only losing a little accuracy -
but for the purpose of this, that didn't matter.

It may be that devices from different manufacturers behave differently?
Mine is made by Allegro.

menotuu, 2024-01-10:
and what version 20A or 30A? I have 30A. 

20A sensitivity typical 100 mV/A
30A sensitivity typical 66 mV/A

SimonRafferty, 2024-01-10:
> <SimonRafferty/Webasto-Heater---Replacement-Controller/issues/11/1884714344

Mine is 20A - but it makes little difference.  It's not being used to
measure an accurate, absolute temperature - just a difference between hot &
cold.  In the code, you can set the flame threshold value - above which it
considers the burner to be hot.  It will need adjusting to suit your sensor.

This is best achieved by looking at a plot of the value as you manually
start & stop the heater.
[image: image.png]
This is mine from Yesterday.  I have a threshold value set as 90 which
seems to work.

closed 2024-01-17 by menotuu