SimonRafferty/Webasto-Heater---Replacement-Controller #12: burning gasoline

menotuu, 2024-01-19:
hi,

sorry for open it as issue. can you convert it as discussion?
I'm trying to run Webastardo on gasoline.

In principle, Webastardo also runs on gasoline with the standard settings for me, but it is far too rich.
So at the moment I have only reduced the consumption, but of course this means that it heats up noticeably slower and too lean is not the best.

When priming and firing up as long as the glow plug is on, I have ~40ppm CO.
The flame comes to light at around 20 seconds.
As soon as the glow plug goes out after 60 seconds, the CO rises rapidly to 320ppm.

I've now gone to the following values, but I can't reduce startfuelcold/warm any further because then the flame goes out. And the fan speed is already at 90% at the factory, so not much more air can get in there.

Or should I reduce the fan speed so that it can be burned longer?

Do you have any tips?

Webasto Thermo Top C Petrol
I dont have a stock webasto petrol pcb. So I can't measure any original values.

my values now:
fuel0.39 fan30 350ppm
fuel0.39 fan90 300ppm

```
//Heater Config 
//*********************************************************************************
int heater_min = 55; // Increase fuel if below
int heater_target = 60; // degrees C Decrease fuel if above, increase if below.
int water_warning = 70;// degrees C - At this temperature, the heater idles
int water_overheat = 85;// degrees C - This is the temperature the heater will shut down
int flame_threshold = 75; //Exhaust temperature above which we assume it's alight

//Fuel Pump Setting
int pump_size = 22; //22,30,60 // läuft vorher mit 22 If the exhaust is consistently smokey, reduce this number  //If you get no fuel (pump not clicking) increase this number

//Fuel Mixture
//If you find the exhaust is smokey, increase the fan or reduce the fuel
float throttling_high_fuel = 1.0;                                                         // stock 1.8        
//float throttling_high_fuel = 1.6; //In summer, exhaust gets too hot on startup
float throttling_high_fan = 90;                                                                 // stock90
float throttling_steady_fuel = 0.9;                                                       // stock1.3
float throttling_steady_fan = 65;                                                               // stock65
float throttling_low_fuel = 0.83;   //(Winter Setting)                                          // stock0.83
float throttling_low_fan = 55;                                                                  // stock55
//Just enough to keep it alight at idle
float throttling_idle_fuel = 0.6; //Do not reduce this value                                    // stock0.6
float throttling_idle_fan = 30;                                                                 // stock30
 
float prime_low_temp = 10;  //Below this temp, prime fueling is increased                       //stock10
float prime_high_temp = 20; //Not used at the moment                                            // stock20
float prime_fan_speed = 15;                                                                     //stock15
float prime_low_temp_fuelrate = 0.8;                                                       //stock 3.5         
float prime_high_temp_fuelrate = 0.5;                                                      //stock2.0

float start_fan_speed = 30;                                                                     // stock30
float start_fuel = 1;             //Summer setting                                              // stock1
float start_fuel_Threshold = -10; //Exhaust temperature, below which to use start_fuel_Cold
float start_fuel_Cold = 0.55;      //Winter Setting (use below 10C)       9                 // stock1.2  
float start_fuel_Warm = 0.39;      //Summer Setting (use above 10C)       // stock1.0   

int full_power_increment_time = 30; // volle Leistungssteigerungszeit in Sekunden

```

SimonRafferty, 2024-01-19:
I wouldn't worry too much about high CO while the heater is starting up.  If you measure the emissions from your car during a cold-start they will be dreadful too!  

Wait until the heater has reached a decent temperature >50C but still running High Fan & Fuel then measure.  Likewise when it steps down the fueling, but is still hot.

menotuu, 2024-01-19:
unfortunately it smells a lot like gasoline too