SimonRafferty/Webasto-Heater---Replacement-Controller #13: Question about compatibility with ESP32

skajer, 2025-01-24:
Hello, very nice project! 

I was wondering if this project would work on an ESP32 microcontroller? If so, what changes would be needed to adapt it for this environment?

Best regards

SimonRafferty, 2025-01-24:
It was originally developed for the Adafruit ESP32 Feather, however, they have an issue with the ADC's sharing a power supply with the WiFi module.  I found that if WiFi had ever been enabled, even if it is currently disabled, the ADC readings were too noisy to be useful!

I have had success using a later ESP32 C3 which does not seem to suffer from these problems.  Unfortunately, that has a completely different pinout so would need a different PCB. 

Another option (possibly better) is to avoid the ADC's altogether by a using separate I2C ADC board such as ADS1115.  It wouldn't be hard to incorporate this. 

I changed to the SAMD21 to avoid this issue as it was almost pin for pin the same as the ESP32 module and I was in a hurry to get it working.  IIRC it only needed a couple of pin assignment changes from the ESP32 version.

I live in my RV during the week as I work a long way from home and it has been running almost every day for 4 years now.

tcrichton, 2025-11-25:
Thank you both.

So if I understood correctly you found too much noise in the ESP32 v4 on the hcalory even with good thermistors but had or heard of success using the original (do you recall if that includes v3 with the flame detection?) on an hcalory.

I'm thinking I either get a used webasto to tinker with or another hcalory so I don't break the current heater prematurely and during winter 😆 
I do run home assistant (house and camper) but happy with an "on/off" trigger if the result is more efficient with less cycles!

Glad work is busy for you, I see you too are in film. You do the real-world magic. I’m in IT for the VFX side, so all my chaos stays inside the computer.

SimonRafferty, 2025-11-25:
I've bought several used Webasto's from BMW E53 X5's which use CANBUS - so not as easy to re-purpose for most people, but ideal for me!  The last one (minus the water & fuel pumps) only cost £20 on eBay!

If you email me (address on IMDB) I'll have a look and see if I have one of the V4 boards looking for a home.  You'd be welcome to it (for free) if you want to tinker some more!